ghoti57/evofw3 #1: BusWare SSC

lustasag, 2020-07-09:
I was wondering if the firmware was compatible with [this module](http://busware.de/tiki-index.php?page=SCC)?

It's a board that sits on top of a RB Pi, compatible with 'culfw' firmware.

[image]

ghoti57, 2020-07-09:
I'm afraid not.  The software will not run on a pi.  It will currently only run on an atmega328.

lustasag, 2020-07-09:
Thanks for responding. The module itself has an ATMEL processor on it.

ghoti57, 2020-07-10:
Sorry, I didn’t look at the details carefully enough and missed the m1284p.

 

I’m not sure if the FW will run on this board.
Three questions need to be answered that aren’t clear the information at the link.

1. Although the cc1101 supports 433 MHz and 868 Mhz there’s a hardware difference between the chip and the antenna – it’s NOT software selectable.  The board MUST be designed for 868MHz operation.

2. The FW requires at least a 16 MHZ clock on the MCU -  8 MHz is not fast enough for the processing required.

 

3. If the board satisfies the first two conditions then the connections from the cc1101 to the m1284p need to be on appropriate pins.
The FW may work as is or may need some minor reassignments.

lustasag, 2020-07-10:
Great, thanks for the info! I'll check out the details next week.

lustasag, 2020-07-16:
The producer informed me that all their MCUs are clocked at 8MHz. So I guess we can close this issue :-(

closed 2020-08-25 by ghoti57