ghoti57/evofw3 #2: Sending commands

dystechnic, 2020-08-25:
Hi,

I'm using evfw3 on a DiY NanoCul and receiving communication from the evohome environment goes well. 
I also tried to send commands but without result. Is sending of commands supported with evofw3? 

Regards,
Ed.

ghoti57, 2020-08-25:
Ed,

I've only just got TX working (this week!) but it's had limited test.

You currently need to use the UART branch, it hasn't been merged to master yet.

Feedback at this stage very welcome

Pete

dystechnic, 2020-08-25:
Thanx.
I will try that branche this weekend. I have a separate Cul for testing purposes only, so it won't interfere with my production environment ;-)

I'll keep you informed of the results.

dystechnic, 2020-08-26:
Hi Pete,

I did some testing with the UART branche tonight. But without success for now.
First I tried to change a setpoint of one of my TRV's. to 15 degrees This was the result:
```
2020-08-26 20:25:23 |1/ - | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command SENT
2020-08-26 20:25:53 |     | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command NOT acknowledged. Resending attempt 1 of 5...
2020-08-26 20:25:53 |1/ - | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command SENT
2020-08-26 20:25:53 |1/000| SETPOINT_OVERRIDE   | W | GWAY EvoGateway        -> 01:139901              | 15.00°C @ Woonkamer               [Zone 1  ]
2020-08-26 20:26:23 |     | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command NOT acknowledged. Resending attempt 2 of 5...
2020-08-26 20:26:23 |1/ - | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command SENT
2020-08-26 20:26:23 |1/000| SETPOINT_OVERRIDE   | W | GWAY EvoGateway        -> 01:139901              | 15.00°C @ Woonkamer               [Zone 1  ]
2020-08-26 20:26:53 |     | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command NOT acknowledged. Resending attempt 3 of 5...
2020-08-26 20:26:53 |1/ - | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command SENT
2020-08-26 20:26:53 |1/000| SETPOINT_OVERRIDE   | W | GWAY EvoGateway        -> 01:139901              | 15.00°C @ Woonkamer               [Zone 1  ]
2020-08-26 20:27:23 |     | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command NOT acknowledged. Resending attempt 4 of 5...
2020-08-26 20:27:23 |1/ - | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command SENT
2020-08-26 20:27:23 |1/000| SETPOINT_OVERRIDE   | W | GWAY EvoGateway        -> 01:139901              | 15.00°C @ Woonkamer               [Zone 1  ]
2020-08-26 20:27:53 |     | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command NOT acknowledged. Resending attempt 5 of 5...
2020-08-26 20:27:53 |1/ - | COMMAND_OUT         | SETPOINT_OVERRIDE ['Woonkamer': 15 deg C] Command SENT
2020-08-26 20:27:53 |1/000| SETPOINT_OVERRIDE   | W | GWAY EvoGateway        -> 01:139901              | 15.00°C @ Woonkamer               [Zone 1  ]
2020-08-26 20:27:53 |     | COMMAND             | ERROR: Previously sent command 'setpoint_override' failed to send. No ack received from controller
```
Then I tried a ping to the controller, and this was the result:
```
2020-08-26 20:30:01 |1/ - | COMMAND_OUT         | PING : Command SENT
2020-08-26 20:30:01 |1/000| DATE_REQUEST        | RQ| GWAY EvoGateway        -> 01:139901              | Ping/Datetime Sync
2020-08-26 20:30:31 |     | COMMAND_OUT         | PING : Command NOT acknowledged. Resending attempt 1 of 5...
2020-08-26 20:30:31 |1/ - | COMMAND_OUT         | PING : Command SENT
2020-08-26 20:30:42 |1/000| DATE_REQUEST        | RQ| GWAY EvoGateway        -> 01:139901              | Ping/Datetime Sync
2020-08-26 20:31:01 |     | COMMAND_OUT         | PING : Command NOT acknowledged. Resending attempt 2 of 5...
2020-08-26 20:31:01 |1/ - | COMMAND_OUT         | PING : Command SENT
2020-08-26 20:31:01 |1/000| DATE_REQUEST        | RQ| GWAY EvoGateway        -> 01:139901              | Ping/Datetime Sync
2020-08-26 20:31:31 |     | COMMAND_OUT         | PING : Command NOT acknowledged. Resending attempt 3 of 5...
2020-08-26 20:31:31 |1/ - | COMMAND_OUT         | PING : Command SENT
2020-08-26 20:31:31 |1/000| DATE_REQUEST        | RQ| GWAY EvoGateway        -> 01:139901              | Ping/Datetime Sync
2020-08-26 20:32:01 |     | COMMAND_OUT         | PING : Command NOT acknowledged. Resending attempt 4 of 5...
2020-08-26 20:32:01 |1/ - | COMMAND_OUT         | PING : Command SENT
2020-08-26 20:32:01 |1/000| DATE_REQUEST        | RQ| GWAY EvoGateway        -> 01:139901              | Ping/Datetime Sync
2020-08-26 20:32:31 |     | COMMAND_OUT         | PING : Command NOT acknowledged. Resending attempt 5 of 5...
2020-08-26 20:32:31 |1/ - | COMMAND_OUT         | PING : Command SENT
2020-08-26 20:32:31 |1/000| DATE_REQUEST        | RQ| GWAY EvoGateway        -> 01:139901              | Ping/Datetime Sync
2020-08-26 20:32:32 |     | COMMAND             | ERROR: Previously sent command 'ping' failed to send. No ack received from controller
```
Thes logs are from the [evo-listener script](https://github.com/smar000/evohome-Listener) I use to read (and write)_ to the evo environment. 
The raw log for the two commands was:
```
2020-08-26 20:25:23 |-| MQTT_SUB           {u'command': u'setpoint_override', u'arguments': {u'setpoint': 15, u'zone_id': 1}}
2020-08-26 20:25:23 |1| COMMAND_OUT        SETPOINT_OVERRIDE: Sending 'W --- 30:071715 01:139901 --:------ 2349 007 0005DC02FFFFFF'
2020-08-26 20:25:53 |1| COMMAND_OUT        SETPOINT_OVERRIDE: Sending 'W --- 30:071715 01:139901 --:------ 2349 007 0005DC02FFFFFF'
2020-08-26 20:25:53 |1| RELAY_HEAT_DEMAND  --- 048  I --- 01:187875 --:------ 01:187875 0008 002 FC00
2020-08-26 20:25:53 |1| SETPOINT_OVERRIDE  --- 000  W --- 30:071715 01:139901 --:------ 2349 007 0005DC02FFFFFF
2020-08-26 20:26:03 |1| ZONE_HEAT_DEMAND   --- 072  I --- 04:024003 --:------ 01:187875 3150 002 0800
2020-08-26 20:26:15 |1| OPENTHERM_TICKER   --- 061  I --- 10:006288 --:------ 10:006288 1FD4 003 0069E1
2020-08-26 20:26:23 |1| COMMAND_OUT        SETPOINT_OVERRIDE: Sending 'W --- 30:071715 01:139901 --:------ 2349 007 0005DC02FFFFFF'
[... cut ...]
2020-08-26 20:34:11 |1| SETPOINT_CTL       --- 048  I --- 01:187875 --:------ 01:187875 2309 030 0001F40101F40201F40301F40401F40501F40601F40701F40801F40901F4
2020-08-26 20:34:11 |1| ZONE_TEMPERATURE   --- 048  I --- 01:187875 --:------ 01:187875 30C9 030 007FFF017FFF027FFF037FFF047FFF05081B0607C907079A087FFF097FFF
```
There is some other communication in that log, but they are not relevant to this issue.
Hope this helps you perfecting the code. If you want me to test something more specific, just let me know.

Regards,
Ed.

ghoti57, 2020-08-26:
Can you ask the Evo listener developer for their view of what's happening. Tell them which evofw3.  I don't use this system and I have problems interpreting the logs.

I'd rather see the raw serial logs from the com port.

ghoti57, 2020-08-26:
From this line (an RX message)

2020-08-26 20:25:53 |1| RELAY_HEAT_DEMAND  --- 048  I --- 01:187875 --:------ 01:187875 0008 002 FC00
 

Your controller appears to be 01:187875

The standard address for the gateway (CUL) device in the existing FW is 18:318170

 

This is a message sent from the gateway device

2020-08-26 20:26:23 |1| SETPOINT_OVERRIDE  --- 000  W --- 30:071715 01:139901 --:------ 2349 007 0005DC02FFFFFF

 

It seems to be trying to send to 01:139901 which isn’t the correct controller address (so nothing will respond)

I do not know what the address 30:071715 represents.

 

The messages you are sending do not appear to have the correct addresses.

 

Pete

ghoti57, 2020-08-26:
Sorry the gateway address is 18:056026

dystechnic, 2020-08-28:
Hi Pete,

I did some more testing, maybe that helps.
First , my setup is as follows:
I have a NanoCul wich I've configured to have the address 18:056026 as you mentioned before. Then there is the evohome controller (the touchscreen thermostate) wich has address 01:187875. 
When I send a ping command to the controller this is what happens on the serial port

```
␑# uart\evofw3 0.4.1
000 RQ --- 18:056026 01:187875 --:------ 313F 001 00
066  I --- 01:187875 --:------ 01:187875 0008 002 FC00
080  I --- 10:006288 --:------ 01:187875 3150 002 FC00
000 RQ --- 18:056026 01:187875 --:------ 313F 001 00
083  I --- 10:006288 --:------ 10:006288 1FD4 003 006D91
065  I --- 01:187875 --:------ 01:187875 3B00 002 FCC8
000 RQ --- 18:056026 01:187875 --:------ 313F 001 00
000 RQ --- 18:056026 01:187875 ----- 313F 001 00
000 RQ --- 18:056026 01:187875 --:------ 313F 001 00
065  I --- 01:187875 --:------ 01:187875 1F09 003 FF0744
066  I --- 01:187875 --:------ 01:187875 2309 030 0001F40101F40201F40301F40401F40501F40601F40701F40801F40901F4
065  I --- 01:187875 --:------ 01:187875 30C9 030 0007CD0107820207810307FC04079105082F06079E07079E080802090820
000 RQ --- 18:056026 01:187875 --:------ 313F 001 00
```
To me it seems that the CUL (18:056026) is now sending to the controller (01:187875). That should be the correct communication. But there is no communication back from the controller to the CUL device, am I understanding this correctly?

I've also asked the author of the evo-listener script to have a look at the problem but have had no reply yet.

dystechnic, 2020-08-30:
HI,

It's working now. Sending is working with the uart branche.
On my side it was a configuration issue and as soon as I had that resolved it started working.
Thank you for your support, and if you need someone to do some testing for you then just let me know.

Regards,
Ed.

ghoti57, 2020-08-30:
Glad it's working.

Just let me know if you see anything that definitely looks like it's my problem.

Peter

closed 2020-09-23 by ghoti57