ghoti57/evofw3 #3: OpenTherm messages on UART branch?

smar000, 2020-09-05:
Hi Pete

Are opentherm messages generally working on your uart branch? I have been running this branch for the last few days, but do not seem to be getting any messages with command 3220 (https://github.com/Evsdd/The-Evohome-Protocol/wiki/3220:-OpenTherm-Message). 

This was working previously using one of your evofw2 branches.

Thanks.

ghoti57, 2020-09-05:
I don't have an OTG so can't directly comment but other users are not saying there's a problem

There's nothing in the code that would cause a specific message to fail.

smar000, 2020-09-05:
Ok thanks for the quick reply. 

I was just looking through my historical logs, and the messages appear to have stopped a couple of weeks ago - so unlikely to be related to my switchover to your `uart` branch firmware.

Thanks anyway.

closed 2020-09-05 by ghoti57