ghoti57/evofw3 #5: no messages

sveco86, 2021-02-03:
I am using nanoCUL (Arduino Nano V3.0 ATMEGA328P + C1101 868 MHz), but no messages are coming even using minicom 
`minicom -D /dev/ttyUSB0`

nanoCUL is plugged to Raspberry Pi 4b.

For flasing fimware I used Arduino IDE, exported binary to Raspberry and successfully flashed using 
`avrdude -v -p atmega328p -c arduino -P /dev/ttyUSB0 -b 57600 -D -U flash:w:/home/pi/evofw3.ino.arduino_standard.hex:i`

dmesg
`[    0.130508] usbcore: registered new interface driver usbfs
[    0.130593] usbcore: registered new interface driver hub
[    0.130728] usbcore: registered new device driver usb
[    0.131103] usb_phy_generic phy: phy supply vcc not found, using dummy regulator
[    1.144221] usbcore: registered new interface driver r8152
[    1.144302] usbcore: registered new interface driver lan78xx
[    1.144378] usbcore: registered new interface driver smsc95xx
[    1.149552] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.04
[    1.149591] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.149627] usb usb1: Product: xHCI Host Controller
[    1.149655] usb usb1: Manufacturer: Linux 5.4.83-v7l+ xhci-hcd
[    1.149684] usb usb1: SerialNumber: 0000:01:00.0
[    1.151575] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.04
[    1.151611] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.151645] usb usb2: Product: xHCI Host Controller
[    1.151673] usb usb2: Manufacturer: Linux 5.4.83-v7l+ xhci-hcd
[    1.151701] usb usb2: SerialNumber: 0000:01:00.0
[    1.154857] usbcore: registered new interface driver uas
[    1.154977] usbcore: registered new interface driver usb-storage
[    1.166346] usbcore: registered new interface driver usbhid
[    1.166373] usbhid: USB HID core driver
[    1.512911] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    1.695619] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
[    1.695659] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    1.695690] usb 1-1: Product: USB2.0 Hub
[    1.843264] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    1.874575] usb 2-1: New USB device found, idVendor=152d, idProduct=1337, bcdDevice= 5.08
[    1.874613] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.874644] usb 2-1: Product: jmicron
[    1.874670] usb 2-1: Manufacturer: jmicron
[    1.874697] usb 2-1: SerialNumber: 7F833EEF5DC0
[    2.023003] usb 1-1.2: new full-speed USB device number 3 using xhci_hcd
[    2.166652] usb 1-1.2: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[    2.166692] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.166727] usb 1-1.2: Product: NANO CUL
[    2.166754] usb 1-1.2: Manufacturer: SHK
[    2.166780] usb 1-1.2: SerialNumber: 868
[    8.009488] usbcore: registered new interface driver usbserial_generic
[    8.009560] usbserial: USB Serial support registered for generic
[    8.075826] usbcore: registered new interface driver ftdi_sio
[    8.075944] usbserial: USB Serial support registered for FTDI USB Serial Device
[    8.076399] usb 1-1.2: Detected FT232RL
[    8.087741] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB0
[    8.251806] usbcore: registered new interface driver brcmfmac`

lsusb
`Bus 002 Device 002: ID 152d:1337 JMicron Technology Corp. / JMicron USA Technology Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

Can you help me with this issue? 

Thanks

ghoti57, 2021-02-03:
Which version of the Arduino GUI are you using?
What Board and options did you use to compile/build?

sveco86, 2021-02-03:
I think it was issue with device. I replaced nanoCUL, flashed same FW and it works.

closed 2021-02-03 by sveco86