ghoti57/evofw3 #6: Setting controller mode or set point not working from Domoticz

FrankvdAa, 2021-02-14:
Hi,

I bought a NanoCUL a few months back to be able to control my Evohome from Domoticz.

Have been able to discover all valve, but I'm not able to set the controller mode (normal, economy, off, ...) or set points.

I do see Domoticz sending the command, but don't see any response to that:

000  W --- 18:003618 01:204708 --:------ 2E04 008 01FFFFFFFFFFFF00
000  W --- 18:003618 01:204708 --:------ 2E04 008 00FFFFFFFFFFFF00

Somebody I know has a real HGI80 and he sees the following when changing controller mode:

095  W --- 18:012786 01:072015 --:------ 2E04 008 01FFFFFFFFFFFF00
049  I --- 01:072015 --:------ 01:072015 2E04 008 01FFFFFFFFFFFF00
047  I --- 01:072015 01:072015 --:------ 2E04 008 01FFFFFFFFFFFF00
095  W --- 18:012786 01:072015 --:------ 2E04 008 00FFFFFFFFFFFF00
049  I --- 01:072015 --:------ 01:072015 2E04 008 00FFFFFFFFFFFF00
047  I --- 01:072015 01:072015 --:------ 2E04 008 00FFFFFFFFFFFF00

Not sure what the three digits at the beginning of each line represent, but I noticed that in my setup it's always 000 when the second column is showing 'W'.

I understood that TX should be working with the latest firmware, or am I wrong?

Please let me know if you need me to run any tests or in case you need more information.

Thanks,

Frank

ghoti57, 2021-02-15:
The 3 digits at the front of the message indicate the signal strength of the received messages (RSSI).  Low numbers are better than high numbers.

The HGI80 prints a non-zero value (always the same) when it TXs a message, Evofw3 always prints zero.  neither value makes more sense than the other because RSSI can only be measured by the RX device.  It does make TX message much more visible in evofw3 though.

ghoti57, 2021-02-15:
Not a bug :-)

ghoti57, 2021-02-15:
What version of evofw3 are you running?

FrankvdAa, 2021-02-15:
Running latest version (v0.6.7) as of yesterday. Updated before running above tests. Haven't seen it working, even with older versions.

Any idea why my Evohome is not responding to those set requests? Where should I start looking?

ghoti57, 2021-02-15:
My first thought is that the nanoCUL is not close enough to the controller.
I'm not aware of any reason why 0.6.7 should not work.

Do you have any logs that include other messages from the controller?

FrankvdAa, 2021-02-15:
Distance between controller and nanoCUL is about 1-1.5m so I would expect that to be okay.

I will add some logs later today.

ghoti57, 2021-02-15:
Logs with timestamps of possible.

FrankvdAa, 2021-02-15:
Here's a log with timestamps:

2021-02-15 11:54:12 >> 035  I --- 01:204708 --:------ 01:204708 1F09 003 FF073A
2021-02-15 11:54:12 >> 035  I --- 01:204708 --:------ 01:204708 2309 027 00076C0105DC02076C0303E80403E80503E80605DC07064008076C
2021-02-15 11:54:12 >> 036  I --- 01:204708 --:------ 01:204708 30C9 027 0007760106CE02074C03061A0405860505BA06062B07061008075F
2021-02-15 11:54:31 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 00
2021-02-15 11:54:32 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 01
2021-02-15 11:54:33 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:54:33 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 02
2021-02-15 11:54:35 >> 035 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:54:35 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 04
2021-02-15 11:54:36 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 05
2021-02-15 11:54:37 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:54:37 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 06
2021-02-15 11:54:38 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 07
2021-02-15 11:54:39 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 08
2021-02-15 11:54:40 >> 000 RQ --- 18:003618 01:204708 --:------ 1260 001 00
2021-02-15 11:55:15 >> 035 RQ --- 01:204708 10:030173 --:------ 22D9 001 00
2021-02-15 11:55:15 >> 064 RP --- 10:030173 01:204708 --:------ 22D9 003 001E82
2021-02-15 11:55:16 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 0000050000
2021-02-15 11:55:16 >> 066 RP --- 10:030173 01:204708 --:------ 3220 005 00C00500FF
2021-02-15 11:55:16 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 0000110000
2021-02-15 11:55:16 >> 067 RP --- 10:030173 01:204708 --:------ 3220 005 00C0110000
2021-02-15 11:55:16 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 0000120000
2021-02-15 11:55:16 >> 067 RP --- 10:030173 01:204708 --:------ 3220 005 00C0120600
2021-02-15 11:55:16 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 0080130000
2021-02-15 11:55:16 >> 066 RP --- 10:030173 01:204708 --:------ 3220 005 00C01347AB
2021-02-15 11:55:16 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 0080190000
2021-02-15 11:55:16 >> 066 RP --- 10:030173 01:204708 --:------ 3220 005 0040194D00
2021-02-15 11:55:17 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 00801A0000
2021-02-15 11:55:17 >> 065 RP --- 10:030173 01:204708 --:------ 3220 005 00C01A47AB
2021-02-15 11:55:17 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 00801C0000
2021-02-15 11:55:17 >> 065 RP --- 10:030173 01:204708 --:------ 3220 005 00C01C47AB
2021-02-15 11:55:17 >> 036 RQ --- 01:204708 10:030173 --:------ 3220 005 0080730000
2021-02-15 11:55:17 >> 066 RP --- 10:030173 01:204708 --:------ 3220 005 00B0730000
2021-02-15 11:55:24 >> 067  I --- 10:030173 --:------ 10:030173 1FD4 003 004828
2021-02-15 11:55:33 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:55:33 >> 067 RP --- 10:030173 01:204708 --:------ 3EF0 009 0000100A00FF034E00
2021-02-15 11:56:33 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:56:33 >> 067 RP --- 10:030173 01:204708 --:------ 3EF0 009 0000100A00FF034E00
2021-02-15 11:56:42 >> 044  I --- 02:004323 --:------ 01:204708 3150 002 0886
2021-02-15 11:56:48 >> 044  I --- 02:004323 --:------ 02:004323 0008 002 FC00
2021-02-15 11:56:48 >> 044  I --- 02:004323 --:------ 02:004323 3150 002 FC86
2021-02-15 11:57:17 >> 036  I --- 01:204708 --:------ 01:204708 1F09 003 FF073A
2021-02-15 11:57:17 >> 036  I --- 01:204708 --:------ 01:204708 2309 027 00076C0105DC02076C0303E80403E80503E80605DC07064008076C
2021-02-15 11:57:17 >> 036  I --- 01:204708 --:------ 01:204708 30C9 027 0007760106CE02074C03061A0405860505BA06062B07061008075F
2021-02-15 11:57:33 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:57:33 >> 067 RP --- 10:030173 01:204708 --:------ 3EF0 009 0000100A00FF034E64
2021-02-15 11:57:49 >> 043  I --- 02:004323 --:------ 02:004323 0008 002 FA00
2021-02-15 11:57:49 >> 043  I --- 02:004323 --:------ 02:004323 22D0 004 00000002
2021-02-15 11:57:57 >> 067  I --- 10:030173 --:------ 10:030173 1FD4 003 004829
2021-02-15 11:58:07 >> 063  I --- 04:009093 --:------ 04:009093 30C9 003 000629
2021-02-15 11:58:31 >> 071  I --- 34:038973 --:------ 34:038973 30C9 003 000760
2021-02-15 11:58:34 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:58:34 >> 067 RP --- 10:030173 01:204708 --:------ 3EF0 009 0000100A00FF034E64
2021-02-15 11:59:31 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 00
2021-02-15 11:59:32 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 01
2021-02-15 11:59:33 >> 075  I --- 04:009067 --:------ 04:009067 30C9 003 000625
2021-02-15 11:59:33 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 02
2021-02-15 11:59:34 >> 035 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:59:34 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 03
2021-02-15 11:59:35 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 04
2021-02-15 11:59:36 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:59:36 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 05
2021-02-15 11:59:37 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 06
2021-02-15 11:59:38 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 11:59:38 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 07
2021-02-15 11:59:39 >> 000 RQ --- 18:003618 01:204708 --:------ 30C9 001 08
[... cut ...]
2021-02-15 12:00:22 >> 036  I --- 01:204708 --:------ 01:204708 30C9 027 0007760106CE02074C03061A0405860505BA060629070625080760
2021-02-15 12:00:22 >> 061  I --- 04:007040 --:------ 01:204708 2309 003 0303E8
2021-02-15 12:00:29 >> 067  I --- 10:030173 --:------ 10:030173 1FD4 003 00482A
2021-02-15 12:00:38 >> 036 RQ --- 01:204708 10:030173 --:------ 3EF0 001 00
2021-02-15 12:00:38 >> 066 RP --- 10:030173 01:204708 --:------ 3EF0 009 0000100A00FF034E64

Just before 12:00:00 you can see me changing the controller mode a few times.

ghoti57, 2021-02-15:
I don't know why this isn't working.  I'm in regular contact with at least one other Domoticz user who is not reporting this problem.

Have you raised this problem on Domoticz or other forums where other users have indicated they're using evofw3?
If so which ones?

FrankvdAa, 2021-02-15:
No, only reported it here because I see the command to set the controller mode coming from Domoticz.

Therefor I thought it had to be related to evofw3 firmware, but as I don't fully know how it's working I might be wrong here ;-)

I guess all other data is only RX which seems to be working fine.

FrankvdAa, 2021-02-15:
I've installed evohome-listener and that is also reporting an error:

2021-02-15 19:51:39 |1/042| ZONE_HEAT_DEMAND    | I | 02:004323              -> CONTROLLER             |   54.0%                           [Zone 9  ]
2021-02-15 19:51:39 |1/042| ZONE_HEAT_DEMAND    | I | UFH 02:004323          -> BROADCAST MESSAGE      |   54.0%                           [Zone 1  ]
2021-02-15 19:51:55 |1/043| SETPOINT_UFH        | I | UFH 02:004323          -> BROADCAST MESSAGE      | 19.00°C                           [Zone 1  ]
2021-02-15 19:52:06 |1/038| SYNC                | I | CONTROLLER             -> BROADCAST MESSAGE      | Next sync at 19:55:11 (in 185.0 secs)
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 19.00°C                           [Zone 1  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 15.00°C                           [Zone 2  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 19.00°C                           [Zone 3  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 10.00°C                           [Zone 4  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 10.00°C                           [Zone 5  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 10.00°C                           [Zone 6  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 15.00°C                           [Zone 7  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 16.00°C                           [Zone 8  ]
2021-02-15 19:52:06 |1/038| SETPOINT_CTL        | I | CONTROLLER             -> BROADCAST MESSAGE      | 19.00°C                           [Zone 9  ]
2021-02-15 19:52:07 |1/037| ZONE_TEMPERATURE    | I | CONTROLLER             -> BROADCAST MESSAGE      | 20.08°C                           [Zone 1  ]
2021-02-15 19:52:07 |     | ERROR               | '1' on line 723 [Command ZONE_TEMPERATURE, data: '--- 037  I --- 01:204708 --:------ 01:204708 30C9 027 0007D80107640207A90306530405C60505FB06064D070621080788', port: 1]
Traceback (most recent call last):
  File "evogateway.py", line 723, in process_received_message
    COMMANDS[msg.command_code](msg)
  File "evogateway.py", line 904, in zone_temperature
    mqtt_publish("{}/{}".format(zones[zone_id], msg.source_name), "temperature",temperature)
KeyError: 1

2021-02-15 19:52:16 |1/042| RELAY_HEAT_DEMAND   | I | UFH 02:004323          -> BROADCAST MESSAGE      |  100.0% @ OTB OpenTherm Bridge
2021-02-15 19:52:16 |1/043| ZONE_HEAT_DEMAND    | I | UFH 02:004323          -> BROADCAST MESSAGE      |   54.0%                           [Zone 252]
2021-02-15 19:52:18 |1/038| ACTUATOR_STATE      | RQ| CTL 01:204708          -> OTB 10:030173          | Request actuator state update
2021-02-15 19:52:18 |1/073| ACTUATOR_STATE      | RP| OTB 10:030173          -> CONTROLLER             |     OFF

Don't know if it is somehow related, but I'll take that up with the maintainer of evohome-listener in parallel.

ghoti57, 2021-02-15:
Please confirm the version of evofw3 by obtaining the first line of the raw output from the nanoCUL after you connect the serial port.  Use the arduino Serial monitor.

FrankvdAa, 2021-02-15:
This is what I get through Arduino Serial Monitor:
>> 20:07:00.421 -> # evofw3 0.6.7

FrankvdAa, 2021-02-15:
Can I somehow send the command to set controller mode through Arduino Serial Monitor?

ghoti57, 2021-02-15:
Cut and paste a message from your log.  Everything after the 000

ghoti57, 2021-02-15:
Make sure you set the serial monitor line ending to CR&NL

ghoti57, 2021-02-15:
FYI evohome_rf is getting far more development effort than evohome_listener

FrankvdAa, 2021-02-15:
Sending below message through Arduino Serial Monitor is showing the same:
W --- 18:003618 01:204708 --:------ 2E04 008 00FFFFFFFFFFFF00
W --- 18:003618 01:204708 --:------ 2E04 008 01FFFFFFFFFFFF00

> FYI evohome_rf is getting far more development effort than evohome_listener

Thanks, I'll look into that also then ;-)

ghoti57, 2021-02-15:
I don't know why this isn't working.  You might ask at evohome_rf if they have ideas.

ghoti57, 2021-02-15:
Where did you buy the nanoCUL

FrankvdAa, 2021-02-15:
Bought it from eBay, German reseller:

FrankvdAa, 2021-02-15:
https://www.ebay.nl/itm/nanoCUL-USB-Stick-FTDI-CC1101-868MHz-FW-1-67-Knick-Antenne-FHEM-CUL-868-Adapter/372221622516?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

FrankvdAa, 2021-02-15:
Beginning to think it might be hardware related.

Now running evohome_rf and every 'send_data' call is being retried 3 times and then I receive a timeout:

FrankvdAa, 2021-02-15:
21:20:44.166 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=1, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:44.266409: RE-SENT (1/3)
21:20:44.204 || HGI:003618 | CTL:204708 | RQ | device_info      | 00       || {}
21:20:44.611 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=2, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:44.810766: RE-SENT (2/3)
21:20:44.636 || HGI:003618 | CTL:204708 | RQ | device_info      | 00       || {}
21:20:45.443 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=3, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:45.843070: RE-SENT (3/3)
21:20:45.485 || HGI:003618 | CTL:204708 | RQ | device_info      | 00       || {}
21:20:47.090 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=3, want=RP|01:204708|10E0|00, tout=2021-02-15 21:20:47.086022: EXPIRED

ghoti57, 2021-02-15:
Do you have a voltmeter.  Can you check connection between Nano D3/D4 and cc1101 GDO0/GDO2

FrankvdAa, 2021-02-15:
> Do you have a voltmeter. Can you check connection between Nano D3/D4 and cc1101 GDO0/GDO2

Do have a volt meter, but nanoCUL is shrinkwrapped so cannot access the tracks with my meter.

My nanoCUL has a circuitboard between Arduino and CC1101.

Am I right when saying that D4 is not used and D3 is connected to GDO2? Can't see where GDO0 is going, also seems not to be connected.

ghoti57, 2021-02-15:
D4 is not connected.  If D3 is connected to GDO2 then D2 should be connected to GDO0.
Can you pierce wrap with needles.  If so try to test connection on cc1101 board not on daughter board.

ghoti57, 2021-02-15:
Forget tracks just go for nano pins and cc1101 connections

FrankvdAa, 2021-02-15:
Measures 20-40mV between GPO0/GPO2 and D2/D3

ghoti57, 2021-02-15:
I don't actually want volts.  I need you to test connectivity between D2-GDO0 and D3-GDO2  ( might have D2/D3 swapped)

ghoti57, 2021-02-15:
If there's a hardware problem the most likely fault is that GDO0 is not connected properly

ghoti57, 2021-02-15:
Just checked and D3 should be connected to GDO0.  Since RX is working D2 must be connected to GDO2.

FrankvdAa, 2021-02-15:
> I don't actually want volts. I need you to test connectivity between D2-GDO0 and D3-GDO2 ( might have D2/D3 swapped)

Thought you wanted volts as you asked for a voltmeter :D

Just checked and D2 is connected to GPO2, D3 is connected to GPO0.

ghoti57, 2021-02-15:
What options in the Arduino IDE did you sue to build the firmware?

FrankvdAa, 2021-02-15:
Do you mean board settings? Tried two options:

1. Arduino AVR boards --> Arduino NANO
   Processor: ATmega328P (old bootloader)

2. evofw3 AVR boards --> atmega328p (SW UART)
   Processor: atmega328P (5V, 16MHz)
   Pinout: Nano
   Bootloader: Old Bootloader

Then did a verify/compile and upload.

iMiMx, 2021-02-15:
Just to add, I suspect I have the same problem - using a nanoCUL from the same eBay seller, with evohome_cc with Home Assistant.  I can see all the various values, but I just tried to manually change the temperature in zone and nothing appeared to change.

Will do some more digging.

ghoti57, 2021-02-15:
Both build options should work but stick to the evofw3 avr boards option.

If you have two nanoCULs what does the second one report when you TX from the first?
In the 'listening" one type !T3 in serial monitor.

ghoti57, 2021-02-15:
Sorry - didn't spot the second contributor there.
Do either of you have two devices?

FrankvdAa, 2021-02-15:
Unfortunately I only have one, so can't test that.

@iMiMx awaiting your results, sounds like there's something different with these nanoCULs.

iMiMx, 2021-02-15:
I also tried the 2 board settings you mention above - without the custom board and with.

On the evohome_cc thread, in the Home Assistant forums, my zones took some time for the names to be retrieved - the dev there says this potentially suggests its not sending:

```
evohome_rf uses three distinct means for collecting data, such as zone temps, etc.

schema file (broken at the moment)
eavesdropping (it captures the packets between the controller and other devices)
discovery (by sending a packet to a controller asking a quesion)
For a typical system, discovery takes 2 minutes (not, say, 47 minutes).

If you don’t any zone names (discovery isn’t working) and a name appears when you push the button on a TRV (specifically a HR82UK) (eavesdropping is working), then your evofw3/nanoCUL is not sending packets & you need to check your settings when you flash the firmware.
```

ghoti57, 2021-02-15:
I'm not aware of anyone who's using one of these devices.

FrankvdAa, 2021-02-15:
Same here, took some time for all devices to be discovered by Domoticz.

ghoti57, 2021-02-15:
do either of you have a logic analyser?

iMiMx, 2021-02-15:
```
2021-02-15 21:30:16 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:215029|3EF1|00): boff=1, want=RQ|13:215029|3EF1|00, tout=2021-02-15 21:30:16.528318: RE-SENT (1/1)
2021-02-15 21:30:17 ERROR (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:215029|3EF1|00): boff=1, want=RP|13:215029|3EF1|00, tout=2021-02-15 21:30:17.126126: EXPIRED
2021-02-15 21:30:17 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:246213|0008|00): boff=1, want=RQ|13:246213|0008|00, tout=2021-02-15 21:30:17.459575: RE-SENT (1/1)
2021-02-15 21:30:17 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:246213|3EF1|00): boff=1, want=RQ|13:246213|3EF1|00, tout=2021-02-15 21:30:17.888515: RE-SENT (1/1)
2021-02-15 21:30:17 ERROR (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:246213|3EF1|00): boff=1, want=RQ|13:246213|3EF1|00, tout=2021-02-15 21:30:17.888515: EXPIRED
2021-02-15 21:31:06 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:246213|0008|00): boff=1, want=RQ|13:246213|0008|00, tout=2021-02-15 21:31:07.068531: RE-SENT (1/1)
2021-02-15 21:31:07 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:246213|3EF1|00): boff=1, want=RQ|13:246213|3EF1|00, tout=2021-02-15 21:31:07.336557: RE-SENT (1/1)
```

iMiMx, 2021-02-15:
> do either of you have a logic analyser?

Afraid not.

FrankvdAa, 2021-02-15:
> ```
> 2021-02-15 21:30:16 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:215029|3EF1|00): boff=1, want=RQ|13:215029|3EF1|00, tout=2021-02-15 21:30:16.528318: RE-SENT (1/1) [...]

Very similar to what I'm seeing with evohome_rf; in my case it tries 3 times and then expires.

FrankvdAa, 2021-02-15:
> do either of you have a logic analyser?

Sorry, I don't. Maybe a colleague of mine has, can ask him tomorrow.

ghoti57, 2021-02-15:
I'd trust zxdavb's interpretation.  It really does seem like there's no TX working.

zxdavb, 2021-02-15:
All - be careful - there is a different between *all* packets EXPIRED, and _some_ packets EXPIRED.

FrankvdAa, 2021-02-15:
> All - be careful - there is a different between _all_ packets EXPIRED, and _some_ packets EXPIRED. Please provide a HA log from the start - grep 'evohome'

No HA here, but evohome_rf... would that be fine too?

FrankvdAa, 2021-02-15:
Snippet of a log from earlier today:

21:20:17.898 # evohome_rf 0.5.5
21:20:17.951 PktProtocolQos.send_data( I|63:262142|0001|00): boff=0, want= I|63:262142|0001|00, tout=2021-02-15 21:20:18.001519: RE-SENT (1/24)
21:20:18.004 PktProtocolQos.send_data( I|63:262142|0001|00): boff=0, want= I|63:262142|0001|00, tout=2021-02-15 21:20:18.054074: RE-SENT (2/24)
21:20:18.056 PktProtocolQos.send_data( I|63:262142|0001|00): boff=0, want= I|63:262142|0001|00, tout=2021-02-15 21:20:18.106414: RE-SENT (3/24)
21:20:18.109 PktProtocolQos.send_data( I|63:262142|0001|00): boff=0, want= I|63:262142|0001|00, tout=2021-02-15 21:20:18.158841: RE-SENT (4/24)
21:20:18.161 PktProtocolQos.send_data( I|63:262142|0001|00): boff=0, want= I|63:262142|0001|00, tout=2021-02-15 21:20:18.211265: RE-SENT (5/24)
21:20:18.181 # evofw3 0.6.7
21:20:18.213 PktProtocolQos.send_data( I|63:262142|0001|00): boff=0, want= I|63:262142|0001|00, tout=2021-02-15 21:20:18.263730: RE-SENT (6/24)
21:20:18.256 || HGI:003618 | NUL:262142 |  I | rf_unknown       | 00FFF... || {'unknown_0': 'FFFF', 'unknown_1': '02', 'unknown_2': '00'}
21:20:43.900 || CTL:204708 | OTB:030173 | RQ | actuator_state   | 00       || {}
21:20:43.932 || OTB:030173 | CTL:204708 | RP | actuator_state   | 00001... || {'actuator_enabled': False, 'modulation_level': 0.0, '_unknown_0': '10', 'flame_active': True, 'flame_state': '0A', '_unknown_1': '00FF033D64'}
21:20:43.965 || HGI:003618 | CTL:204708 | RQ | device_info      | 00       || {}
21:20:44.166 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=1, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:44.266409: RE-SENT (1/3)
21:20:44.204 || HGI:003618 | CTL:204708 | RQ | device_info      | 00       || {}
21:20:44.611 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=2, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:44.810766: RE-SENT (2/3)
21:20:44.636 || HGI:003618 | CTL:204708 | RQ | device_info      | 00       || {}
21:20:45.443 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=3, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:45.843070: RE-SENT (3/3)
21:20:45.485 || HGI:003618 | CTL:204708 | RQ | device_info      | 00       || {}
21:20:47.090 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=3, want=RP|01:204708|10E0|00, tout=2021-02-15 21:20:47.086022: EXPIRED
21:20:47.132 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000D     || {'domain_id': 'FA', 'device_class': 'hotwater_sensor'}
21:20:47.337 PktProtocolQos.send_data(RQ|01:204708|000C|000D): boff=1, want=RQ|01:204708|000C|000D, tout=2021-02-15 21:20:47.437286: RE-SENT (1/3)
21:20:47.372 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000D     || {'domain_id': 'FA', 'device_class': 'hotwater_sensor'}
21:20:47.775 PktProtocolQos.send_data(RQ|01:204708|000C|000D): boff=2, want=RQ|01:204708|000C|000D, tout=2021-02-15 21:20:47.975276: RE-SENT (2/3)
21:20:47.803 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000D     || {'domain_id': 'FA', 'device_class': 'hotwater_sensor'}
21:20:48.609 PktProtocolQos.send_data(RQ|01:204708|000C|000D): boff=3, want=RQ|01:204708|000C|000D, tout=2021-02-15 21:20:49.008708: RE-SENT (3/3)
21:20:48.635 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000D     || {'domain_id': 'FA', 'device_class': 'hotwater_sensor'}
21:20:50.241 PktProtocolQos.send_data(RQ|01:204708|000C|000D): boff=3, want=RP|01:204708|000C|000D, tout=2021-02-15 21:20:50.236423: EXPIRED
21:20:50.283 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000F     || {'domain_id': 'FC', 'device_class': 'heating_control'}
21:20:50.488 PktProtocolQos.send_data(RQ|01:204708|000C|000F): boff=1, want=RQ|01:204708|000C|000F, tout=2021-02-15 21:20:50.588503: RE-SENT (1/3)
21:20:50.523 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000F     || {'domain_id': 'FC', 'device_class': 'heating_control'}
21:20:50.928 PktProtocolQos.send_data(RQ|01:204708|000C|000F): boff=2, want=RQ|01:204708|000C|000F, tout=2021-02-15 21:20:51.128277: RE-SENT (2/3)
21:20:50.955 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000F     || {'domain_id': 'FC', 'device_class': 'heating_control'}
21:20:51.758 PktProtocolQos.send_data(RQ|01:204708|000C|000F): boff=3, want=RQ|01:204708|000C|000F, tout=2021-02-15 21:20:52.158372: RE-SENT (3/3)
21:20:51.787 || HGI:003618 | CTL:204708 | RQ | zone_devices     | 000F     || {'domain_id': 'FC', 'device_class': 'heating_control'}
21:20:52.187 || STA:038973 |            |  I | temperature      | 000782   || {'temperature': 19.22}
21:20:53.790 PktProtocolQos.send_data(RQ|01:204708|000C|000F): boff=3, want=RP|01:204708|000C|000F, tout=2021-02-15 21:20:53.788475: EXPIRED
21:20:53.819 || HGI:003618 | CTL:204708 | RQ | tpi_params       | FC       || {'domain_id': 'FC'}
21:20:54.024 PktProtocolQos.send_data(RQ|01:204708|1100|FC): boff=1, want=RQ|01:204708|1100|FC, tout=2021-02-15 21:20:54.123907: RE-SENT (1/3)
21:20:54.058 || HGI:003618 | CTL:204708 | RQ | tpi_params       | FC       || {'domain_id': 'FC'}
21:20:54.464 PktProtocolQos.send_data(RQ|01:204708|1100|FC): boff=2, want=RQ|01:204708|1100|FC, tout=2021-02-15 21:20:54.664364: RE-SENT (2/3)
21:20:54.491 || HGI:003618 | CTL:204708 | RQ | tpi_params       | FC       || {'domain_id': 'FC'}
21:20:55.296 PktProtocolQos.send_data(RQ|01:204708|1100|FC): boff=3, want=RQ|01:204708|1100|FC, tout=2021-02-15 21:20:55.696537: RE-SENT (3/3)
21:20:55.322 || HGI:003618 | CTL:204708 | RQ | tpi_params       | FC       || {'domain_id': 'FC'}
[... cut ...]
21:21:00.064 PktProtocolQos.send_data(RQ|01:204708|313F|00): boff=3, want=RP|01:204708|313F|00, tout=2021-02-15 21:21:00.059011: EXPIRED
21:21:00.105 || HGI:003618 | CTL:204708 | RQ | system_zones     | 0008     || {'zone_type': 'radiator_valve'}
21:21:00.309 PktProtocolQos.send_data(RQ|01:204708|0005|0008): boff=1, want=RQ|01:204708|0005|0008, tout=2021-02-15 21:21:00.409488: RE-SENT (1/3)
21:21:00.346 || HGI:003618 | CTL:204708 | RQ | system_zones     | 0008     || {'zone_type': 'radiator_valve'}

ghoti57, 2021-02-15:
If either of you can get an analyser a pulseview trace of GDO0/GDO2 + MOSI/SCK/MISO/CSN will help.
Sample at 1MHZ.  collect as many samples as possible.
Start the capture and then press the nano reset.
Try to send something (from serial monitor) as soon as possible.
Stop capture

iMiMx, 2021-02-15:
[evo_HA.log](https://github.com/ghoti57/evofw3/files/5984251/evo_HA.log)

ghoti57, 2021-02-15:
@zxdavb I don't think TX is working.
Are you aware of anyone who's used this module successfully?

danrp-git, 2021-02-15:
Having the same problem, also brought via eBay German reseller.

zxdavb, 2021-02-15:
This is definitely a problem - looks like to echo from **evofw3**:
```
21:20:44.166 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=1, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:44.266409: RE-SENT (1/3)
21:20:44.204 || HGI:003618 | CTL:204708 | RQ | device_info | 00 || {}
21:20:44.611 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=2, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:44.810766: RE-SENT (2/3)
21:20:44.636 || HGI:003618 | CTL:204708 | RQ | device_info | 00 || {}
21:20:45.443 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=3, want=RQ|01:204708|10E0|00, tout=2021-02-15 21:20:45.843070: RE-SENT (3/3)
21:20:45.485 || HGI:003618 | CTL:204708 | RQ | device_info | 00 || {}
21:20:47.090 PktProtocolQos.send_data(RQ|01:204708|10E0|00): boff=3, want=RP|01:204708|10E0|00, tout=2021-02-15 21:20:47.086022: EXPIRED
```

danrp-git, 2021-02-15:
2021-02-15 21:30:06 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:045348|1100|00): boff=1, want=RQ|13:045348|1100|00, tout=2021-02-15 21:30:06.619783: RE-SENT (1/3)
2021-02-15 21:30:06 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:045348|1100|00): boff=2, want=RQ|13:045348|1100|00, tout=2021-02-15 21:30:07.147745: RE-SENT (2/3)
2021-02-15 21:30:07 WARNING (MainThread) [evohome_rf.transport] PktProtocolQos.send_data(RQ|13:045348|1100|00): boff=3, want=RQ|13:045348|1100|00, tout=2021-02-15 21:30:08.178980: RE-SENT (3/3)

ghoti57, 2021-02-15:
@danrp-git smart-home-komponente?

danrp-git, 2021-02-15:
Yes

zxdavb, 2021-02-15:
What do you mean by 'this module' - I do not usually pay attention to HW - I have people who do that for me. :)

[143 more replies]

closed 2021-03-21 by ghoti57