ghoti57/evofw3 #7: Afraid I am not very technical with hardware, but if it helps I can send the device to a UK address if anyone wants to investigate further.  Anyway amazing work guys.

ghoti57, 2021-02-15:
Afraid I am not very technical with hardware, but if it helps I can send the device to a UK address if anyone wants to investigate further.  Anyway amazing work guys.

_Originally posted by @domb80 in https://github.com/ghoti57/evofw3/issues/6#issuecomment-779469797_

what's your email address to discuss this offline

domb80, 2021-02-15:
Thanks for getting back to me, is there anyway to respond in private?  Not sure I want to post the address in public here...

ghoti57, 2021-02-15:
No github doesn't have PMs.
I think I can hide messages.
Can you still see your reply above?

ghoti57, 2021-02-15:
Can you make it visible

ghoti57, 2021-02-15:
Do you have a way to PM zxdavb

domb80, 2021-02-15:
Looks like I can edit it too... Hang I am going to do make a temp email addy for this - have a domain.

ghoti57, 2021-02-15:
If you can PM David ask him to send me your address.

ghoti57, 2021-02-15:
Can you see the deleted message?

domb80, 2021-02-15:
use evo@bodger.org.uk and ill reply with the real address and remove the message.  Badger not included