ghoti57/evofw3 #10: can't program this onto my UNO

GlennWardle, 2021-03-11:
I can't upload this to my standard Arduino UNO (ATMEGA328P). If I set my board option to atmega328P (SWUART) It fails to upload immediately saying #error no baud rate defined. If I set the board type to atmega32U4 (HW UART), it gets further, but comes up with: Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding. This error gets repeated multiple times. My programmer is set to AVRISP - although changing that makes no difference. Please - have you any ideas?

FrankvdAa, 2021-03-11:
Did you use the evofw3 AVR board definitions as stated in the README?

ghoti57, 2021-03-12:
And you need to make sure you're using the latest version of the evofw3 board definitions.

You must use an atmega328 or atmega32u4 definition according to your hardware type.

closed 2021-03-12 by ghoti57