ghoti57/evofw3 #12: No response on specific command

reneklootwijk, 2021-03-26:
When I issue the 0x10E0 request targeted to a specific device to an Arduino Nano, running the evofw3 firmware (evofw3\uart 0.6.1), I get no response (even when retrying 10 times). To other devices I do get a response. However when I issue the exact same request via an HGI80 to that specific device I do get a response.

The request that fails to get a response is:

```
RQ --- 18:000730 10:014494 --:------ 10E0 001 00\r
```

When I monitor the request being sent by the evofw3 firmware using an HGI80 I see the following:

```
067 RQ --- 18:070150 10:014494 --:------ 10E0 001 00
```

When I issue the exact same request (so using 18:000730 as the source) and monitor the request on the device running the evofw3 firmware, I see:

```
095 RQ --- 18:009964 10:014494 --:------ 10E0 001 00
```

So the only difference seems to be the source address mapping.

When I issue the request via the HGI80, the response (directly after the first request) I get is:

```
067 RP --- 10:014494 18:009964 --:------ 10E0 038 000001C8810B0700FEFFFFFFFFFF140807E35238383230000000000000000000000000000000
```

Do you have any clue why there is no response when using the evofw3 firmware and an CC1101 while there is a response when using a n HGI80?

ghoti57, 2021-03-26:
Update to evofw3 0.7.0

closed 2021-03-26 by ghoti57

reneklootwijk, 2021-03-26:
I did, same behaviour

ghoti57, 2021-03-26:
what device are you using?
Where did you get it?

reneklootwijk, 2021-03-26:
I have tried two different CC1101, one I had laying around and the other I just received from Aliexpress. The CC1101 is connected to an Arduino nano v3. Everything is working alright so far, except for sending the mentioned request to that specific device.

ghoti57, 2021-03-26:
So this is a self build system rather than a package of nano+cc1101 sold as a nanoCUL.

I assume you probably have the green cc1101 modules.
There have been problems with the 26MHz crystal on some of these not being accurate enough.

evofw3 has an autotune function to try to bypass this problem.

Preferably from the Arduino GUI serial monitor, rather than any other serial program try this.  (make sure line ending is set to NL & CR)

Send the following command (including !)
!FT

This will try to find the optimum frequency parameters for your cc1101.
You will see good AND bad received messages reported and lines that look like
!C D 21 65 5A

This will run for about 40 minutes
You can check whether it's complete at any time by sending
!F

It will report the current frequency parameter and, if it's still searching, will report "Tuning"

When it's finished you can try to send your message
RQ --- 18:000730 10:014494 --:------ 10E0 001 00

Assuming it works save the frequency with
!FS

Let me know the final value reported by !F

reneklootwijk, 2021-03-26:
This worked! The final value is F=21659e. It still does not respond after the first request I have to retry a couple of times, but eventually it responds. Thanks!

ghoti57, 2021-03-27:
The OTG is one of the most difficult devices to communicate with.

How consistent is this message at getting a response
RQ - 18:730 1:xxxxxx - 0016 1 0
where 1:xxxxxx is your controller

The F value you report is not as far from ideal as other users have seen.
It could be that your crystal is not as inaccurate or the autotune did't work as well as it could.

If the message above is not consistent it might be worth running autotune again.
Make sure you don't try to send any messages while it's running.

reneklootwijk, 2021-03-27:
The response is repeatedly: 

```
066 RP --- 01:196209 18:070150 --:------ 0016 002 0017
```

reneklootwijk, 2021-03-27:
I ran the autotune process multiple times, the value I am currently using is: F=2165a6

ghoti57, 2021-03-28:
If you got the response to the 0016 every time there's nothing fundamentally wrong with your device.

Autotune has clearly helped.

However I've just noticed something in your message about using the evofw3 and HGI80 to monitor each other.

The first number in each message is a signal strength indication and while I don't know how the evofw3 number compares to the HGI80 I do know that 095 on the evofw3 is a VERY weak signal.

The evofw3 device is not physically close to the HGI80.  This suggests that they're not the same distance from the OTG.  Comparisons between HGI80 and evofw3 performance under these conditions are not useful.

If they are close together then you probably have a problem with the evofw3 antenna.