ghoti57/evofw3 #22: flashing to nanocul not working

kdw2060, 2021-11-05:
Hi,

I bought the NanoculV3 board linked to under Hardware UART [here in the wiki](https://github.com/ghoti57/evofw3/wiki/Hardware-Platforms#hardware-platforms), then followed the Arduino-IDE instructions to flash the firmware.

However I always get these errors in the Arduino IDE:

```
Arduino: 1.8.16 (Linux), Board:"atmega32u4 (HW UART), atmega32u4 (3.3V 8 MHz), SHA atmega32u4"

De schets gebruikt 18534 bytes (64%)  programma-opslagruimte. Maximum is 28672 bytes.
Globale variabelen gebruiken 1525 bytes (59%) van het dynamisch geheugen. Resteren 1035 bytes voor lokale variabelen. Maximum is 2560 bytes.

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = ""; type = 
 Software Version = 
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
avrdude: error: programmer did not respond to command: exit bootloader
```

I played around with the settings for 'pinout', 'processor' and 'programmer', but the result is always the same.

Anybody has an idea how to fix this?
Thanks in advance.

ghoti57, 2021-11-06:
You refer to the device as a nanoculV3.  Did this device come loaded with CUL firmware?

If so the chances are it has a bootloader that is incompatible with the Arduino GUI and avrdude.

I'll try to find some information about how to load evofw3 on this device

ghoti57, 2021-11-06:
It probably has the DFU bootloader.

See this page
https://www.smarthome-agentur.de/blog/der-neue-cul-stick/

ghoti57, 2021-11-06:
It's also possible that the connections between the 32u4 and cc1101 are not the same as the firmware expects.  That can be resolved if you can get the firmware loaded.

kdw2060, 2021-11-06:
yes, i also found that information, it has the dfu bootloader and culfw installed. I presumed it was colpatible with evofw3 because of the link in the wiki. Is it possible to flash evofw3 with the dfu tool? I see this requires a hex file, is that something that can be compiled from the arduino files?

Thank you for your quick response btw, much appreciated.

brucemiranda, 2021-11-07:
Check what version of CULFW is on the NanoCUL. See https://www.smarthome-agentur.de/blog/tutorial-nanocul-stick-testen-debugen/ download the HTerm software, use the configuration file and then send a "V". 
Then you need the DFU-Flash tool https://www.smarthome-agentur.de/wp-content/download/DFUGUI.zip also see https://www.smarthome-agentur.de/blog/der-neue-cul-stick/
 
You then compile the evofw3 in Arduino with the 32u4 settings and point the DFU-Flash tool at the .HEX file that Arduino generates. 

Assuming you have CULFW on the NanoCUL, on HTerm issue the "B" command, this will put the NanoCUL into bootloader mode. 

Then using the DFU tool perform an Erase. then load the HEX generated by Arduino. 

Once you have evofw3 loaded on to the NanoCUL, it too has a DFU bootloaded mode and uses the same "B" command, that you send via the terminal software (e.g. HTerm)

ghoti57, 2021-11-07:
Thanks Bruce.

btw The evofw3 boot command is actually !B

brucemiranda, 2021-11-07:
While I remember, the Board settings for the CULV3 hardware in Arduino are  "HW UART", "3.3V 8MHz" and "SHA atmega32u4"

kdw2060, 2021-11-07:
thank you Bruce, those steps are getting me further. Not fully successful however. I can put the stick in bootloader mode, but then 'm stuck at the erasing with DFUGUI. In the windows command line feedback i can see 'dfu-programmer: no device present.

btw, the arduino compilation gives you 2 hex files, one with and one without bootloader. I presume i need to use the one without bootloader in the next step.

brucemiranda, 2021-11-07:
Look at the Windows Device Manager. When you first plug the CULV3 in you will get a normal COM port allocated. wWhen you put the CULV3 into boot loader mode you should see that COM disappear and another ATMEL device appear. If not, then you might not have the ATMEL device drivers loaded. You will need them otherwise in bootloader mode, the DFU tool won't find your CULV3. 
Also yes, you should ALWAYS use the Arduino Hex without the bootloader.

kdw2060, 2021-11-08:
ok, think I got it going, seeing the expected packets in packets log of evohome_cc, now figuring out how to actually send commands.
Thanks a lot guys, closing this issue.

closed 2021-11-08 by kdw2060

PAGR-coder, 2021-11-08:
Thx a lot, it helped me as well