ghoti57/evofw3 #25: nano CUL not working

jaccol, 2021-12-01:
I bought a nano CUL, but after loading this software nothing happens.

I suspect that the pinout of the device is different from the pinout that is currently supported.

The device I have is this one: https://www.smart-home-komponente.de/nano-cul/nano-cul-868-extern/
I *guess* - on inspection - that the cc1101 is connected to 3.3V, while there is only a board definition for a atmega328p w/ 5V.

I followed the instructions from the README, pointed the arduino UI to `evofw3.ino` and pushed 'Upload'. There were no errors in that step.

From here I expected the 'Serial Monitor' to show some output. nothing. I tried giving commands like `!F`, `V`, `!V`, etc but there never was any response from the Arduino.

Can anyone give me some pointers on how to proceed debugging?

Thanks,

Jacco

ghoti57, 2021-12-01:
The 5v is the power to the processor, not the cc1101.

It's usb powered so probably 5v. You just need to make sure there's not a regulator reducing the cpu input power to 3.3v

Might be worth asking Smart Home Komponente about schematics

ghoti57, 2021-12-01:
There are alternative build options for the 328 that are different pinouts to the cc1101

jaccol, 2021-12-01:
There is a pdf detailing which pins are used:
https://www.smart-home-komponente.de/support/aufbauanleitungen/

But if I read your board definition, the main question is the order of INT0/INT1
could I try a 'pro' layout in stead of a 'nano' or are there other differences between those?

ghoti57, 2021-12-02:
That's the only difference between the pro and the nano

benjaminvdb, 2022-01-03:
@jaccol Were you able to get the firmware working? I'm considering to purchase the same USB dongle, since the one referenced in this repository is out-of-stock.

jaccol, 2022-01-03:
I have not been able to get it to work, but I also did not spend much time on it, busy with other projects. I won't have physical access to the device for the next three weeks ...

dennisvbussel, 2022-02-16:
FYI i build a custom dongles using the Arduino Pro Micro 3.3V 8Mhz and a 868 radio. It only required a minor rework.

[image]

ghoti57, 2022-02-16:
You appear to have moved the connection for the SPI CS pin,
What build options did you use to build?

The standard options available for 
**Atmega32u4 (3V3 8MHz)** , pinout **Pro Micro** 
should have compiled the firmware to use pin 18 ( Port B pin 6) on your board.

I suspect you used the **SHA atmega32u4** pinout option which does attempt to use Port B Pin 0 for CS.
This pin does not appear on an edge connector which is why you appear to needed to solder on a wire.

ghoti57, 2022-02-16:
@dennisvbussel 

You might want to look at this picture.

[image]

I found a lot of variation in the antennas I used early on.
I found the ones that had a much longer effective length were not working as well.

The one at the top was by far the best, so I experimented with trimming one of the longer ones and improved it to the same as the top one.

You can see that I progressively trimmed more away each time before getting the best result.

The length of the wire in the top antenna was 137mm,  By the time I finished  the bottom one it was about 143mm.
The antenna in your picture looks significantly longer.

When you calculate the length (_L_) you need to take into account the coil diameter(_D_), wire diameter(_d_), number of coils(_n_) and stub length(_l_).

_L = l + n x pi x ( D - d )_

You might want to play with the length of your antenna.

For reference, the antenna in this image works very well.
<https://www.amazon.co.uk/dp/B07NJV1F6F/>
You can see it's very similar to the top one above

dennisvbussel, 2022-02-16:
Yes indeed @ghoti57, I used the default build. Probably it would have been easier to just update te IO config however, now I get a cool looking led blink whenever the SPI bus is addressed :).

Regarding the antenna, I bought and assembled it as-is and so far it is working perfectly within my entire house :).

Antenna theory is cool but can be tricky to tune without proper equipment.

ghoti57, 2023-05-12:
Used incorrect configuration resulting in need to patch connections

closed 2023-05-12 by ghoti57