ghoti57/evofw3 #26: Starting tuning process on Raspberry Pi

Maikel-K, 2021-12-08:
Hi, im trying to check if my setup is working.
I was guessing to check if the autotune process could be run on my Pi.

i tried screen /dev/ttyUSB0 115200 as mentioned in this post:
https://github.com/ghoti57/evofw3/issues/6#issuecomment-783220165

But i doesnt work.
Any tips if this is possible to run on a RPI? If you need any info from me please let me know.
Im not the biggest expert on all of this but im eager to learn and fixed a lot of other things on the RPI.

ghoti57, 2021-12-09:
The autotune process runs on the device not on the pi.
The pi just provides a serial terminal to communicate with the device and trigger autotune.

Why do you want to run autotune?
What device do you have?
What doesn't work?
Are you sure the device is connected to the pi as /dev/ttyUSB0

Maikel-K, 2021-12-09:
Yes it got connected as ttyUSB0.

I want to run the autotune for two reasons:
Check that the communication between the unit and the pi is working

And

As mentioned in the above mentioned linked comment his advise was to run the autotune in the situation the unit will be running.

Or doesnt the autotune make an sense if it has been run already on a whole different location?

ghoti57, 2021-12-10:
That is reporting that the device is receiving messages and reporting them to the pi over the USB connection.  It appears to be
functioning.

I repeat my question - what type of device do you have? Where did you get it?

Autotune is only necessary for devices that have been built with poor quality components.  It won't do anything useful if the device is well made.

What type of system are you trying to monitor?

Maikel-K, 2021-12-10:
Im trying to monitor my Evohome system. If got the first gen color screen with the RFG100 internet gateway and that gateway is failing some times plus I can't see which room is requesting heat. So especially for the heatrequest I want to use this setup.

Im using these components.
[image]
[image]
[image]

It's good to know that the module is working, I now know that I can try and use the Evogateway project of github user smar000. That won't put out data yet and I was wondering if that is because my reciever but that seems to be working/putting data on the serial at least.

ghoti57, 2021-12-10:
So you have a self build device.

Autotune was written specifically to deal with some usb nanoCul devices that used poor quality components on the radio module.

You can run autotune but it probably won't make a significant difference.

Maikel-K, 2021-12-10:
Yes it's a DIY indeed.

It's for the Crystal on the Nano or on the CC1101?

ghoti57, 2021-12-10:
The problem was with the crystal on the cc1101.

Maikel-K, 2021-12-10:
And the tune is saved on the Nano or on the chip of the CC1101?

ghoti57, 2021-12-10:
It's saved on  the nano.  The cc1101 had no non-volatile storage.

Maikel-K, 2021-12-10:
Okay, should it make any difference in which location you run the tune?

Maikel-K, 2021-12-11:
I have run the autotune a few times now on the Pi and got following outcome:
!F F=216605

With this info saved to the unit with !FS im recieving good packages now.
So running it in the place where you are going to run it seems to make sense??

Maikel-K, 2021-12-12:
By the way, are there any benefits of using the Atmega32U4 instead of the Atmega328p?

ghoti57, 2021-12-13:
The 32u4 uses a hardware UART to connect to the CC1101 and has better RX performance than the 328p which uses a software UART.  TX performance should be the same.

Maikel-K, 2021-12-13:
So if I want to send command trough the module it would work better with the 32U4, correct?

Would this unit work?

https://a.aliexpress.com/_m0k2pMw

Or this one:

https://a.aliexpress.com/_mPX8Vaq

Maikel-K, 2022-01-09:
Hi Peter,

Would you suggest any of the two above units or do you have another suggestion?

Any suggestion for the radio C1101 board or keep the one I already have because it's working?

A friend of mine also wants a unit so I was thinking going for the 32U4 or sticking with the 328P which I'm using too.

ghoti57, 2022-01-09:
I will release a batch of these in the next week.

https://indalo-tech.onlineweb.shop/SSM-D/p7844707_21129532.aspx

If you really want to continue with a self build device a 32u4 device is much better. Any 328p device with as usb interface uses the HW UART for the host usb interface.  On the 32u4 it is available for the cc1101 interface and packet RX is much better.

Maikel-K, 2022-01-09:
Hi Peter,

The website says no shipping outside the UK...

So that makes that no option, I'm in the Netherlands.

Which of the two above 32U4 units would you recommend by looking at it?

ghoti57, 2022-01-09:
Check the FAQ about delivery outside the UK which I've recently updated.  I can now shop to the Netherlands but you will have to pay VAT and a handling charge in the Netherlands.

ghoti57, 2022-01-09:
Make sure you request a stock update to be informed about when they are available.

ghoti57, 2023-05-12:
Not really an issue with the code

closed 2023-05-12 by ghoti57