ghoti57/evofw3 #30: No serial communication after successful board programming. nanoCUL board advice needed

johnkingdon, 2022-10-10:
Maybe more hardware related than evofw3, but posting here as I am not sure where else to ask:)

Picked up one of these:
https://www.amazon.de/-/en/gp/product/B07T3H8VF2/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Seems to be the correct radio, frequency and Firmware.
Received yesterday and installed using Arduino IDE (on MacOS) using the following settings:

GetBoard Info returns: atmega328p (SW UART)

So its set up like so:

Tools-> Board "atmega328p (SW UART)"
Tools-> Processor "atmega328p (5V, 16MHz)"
Tools-> Pinout "Nano"
Tools -> Host "115200"
Tools-> Port = "/dev/cu.usbserial-A50285BI"
Tools - Bootloader "Standard Bootloader"

Sketch -> Verify/Compile
Sketch-> Upload

Both complete with no error

In serial monitor of Arduino IDE there is no output
In terminal monitor there is no communication
Using "screen" command in linux there is no output

Any advice on how to see output form the board or to test if it is actually working?
How often do these devices fail?  As Sketch->Upload did not complain, I assume that the board is working.  

I just can't see any output (hence ultimate use with ramses_rf/ramses_cc/HA does not see any output.)

Any advice on how to get this responding?

Many thanks

johnkingdon, 2022-10-10:
FYI Terminal configurations are:

Via Arduino IDE:
/dev/cu.usbserial-A50285BI
Speed 115200

Via Terminal:
/dev/cu.usbserial-A50285BI
Speed 15200
Bits 8
Stop Bits 1
Parity N

similar for screen

ghoti57, 2022-10-10:
Try the alternative Pinout option.  It will switch the two data pins between the 328 and cc1101.

⁣Get BlueMail for Android ​

johnkingdon, 2022-10-10:
Reloaded with Pinout set to "Pro" same result - no communication, see image below.  Also tried "Old Bootloader" and that failed for the upload.  "Standard Bootloader" works (completed upload with no error).

[image]

ghoti57, 2022-10-11:
> <img width="790" alt="Screenshot 2022-10-10 at 19 14 22"

Your not uploading any code that's going to do anything.

You need to get a copy of the code from hitHub/ghoti57/evofw3

⁣Get BlueMail for Android ​

Grish-Lon, 2022-10-11:
Doh -  That would be it then!

More information below for others that may find it useful.

My mistake was thinking that the board definition 

'''
https://raw.githubusercontent.com/ghoti57/evofw3_avr/master/package_evofw3_boards_index.json
'''
took the code as well (as it pointed to a .gz file. (new to Arduino IDE)

So I downloaded evofw3-master and then

1) Opened the evofw3.ino file in Arduino IDE
2) Sketch -> Verify/Compile
3) Sketch -> Upload (tool a little longer this time :)
4) Tools -> Serial Monitor

Now receiving output!

'''
# evofw3 0.7.1
068  I --- 01:073314 --:------ 01:073314 1F09 003 FF050F
068  I --- 01:073314 --:------ 01:073314 30C9 027 0007000106CD0208550307CE0407A705078C0607C107076108067B
067  I --- 01:073314 --:------ 01:073314 0008 002 FA00
080 RQ --- 30:220186 63:262143 --:------ 3EF0 001 00
080 RQ --- 30:220186 63:262143 --:------ 3EF0 001 00
081 RQ --- 30:220186 63:262143 --:------ 3EF0 001 00
080 RQ --- 30:220186 01:073314 --:------ 0006 001 00
065 RP --- 01:073314 30:220186 --:------ 0006 004 00050000
061  I --- 22:042643 --:------ 22:042643 0008 002 0000
062  I --- 22:042643 --:------ 22:042643 0008 002 0000
'''

Next step - work out what's what!

Many thanks for your help,

Grish

Grish-Lon, 2022-10-11:
So it's decoding like a champ now!

How "talkative" is Evohome in general?  I'm getting messages (sometimes a few of them quickly) every few seconds.  Sometimes messages do not come through for up to 20 seconds.  (Roughly 1 out of 10 messages or batches has a delay up to roughly 20 seconds).  In general its 6 to 20 messages per minute. The heating is turned off now.

I may be missing messages, which is not a problem and I understand this will happen.

My question:

Is this "normal" traffic, or should I consider running the tuning routine for the C1101 (as shown in the wiki?

My thinking is this traffic is pretty good, but with no experience with this I thought I'd ask.

Many thanks for the code!

Grish

ghoti57, 2022-10-11:
If you haven't done so jet check out the evohome_cc, evohome_rf  links on  this page
<https://indalo-tech.onlineweb.shop/SSM-D2/p7844707_21584696.aspx>
The dev has a lot of background information in his github wikis

The tuning algorithm was only created to deal with a batch of particularly dodgy cc1101 modules that appeared in late 2020.
They had inaccurate crystals. I can't remember who was actually selling them (but they were from Germany).  
You don't need to run it unless you really think there's a lot of missing messages or TX is not working very well.

The atm328p devices are fundamentally poorer in RX of messages than atm32u4 based devices - they will drop more messages.  This is because they don't have an available HW UART to interface to the cc1101.  Both families of device should TX equally well subject to having an accurate cc1101 crystal.

The number of messages depends on the number of devices in your system.
You'll see a burst of activity every 3-4 minutes (random interval set by controller) where battery operated devices get updated and respond.

Mains powered devices (BDR91 etc) are not tied to this time interval.

ghoti57, 2022-10-11:
I've looked back at the issue that led to me writing autotune.
The devices with the faulty crystal came from the same place as your device.

The crystal has most impact on TX.
You can test this by sending a message to the controller from the Arduino Serial Monitor.

make sure you have line ending set to CR+LF and then try to send this command
RQ --- 18:730 01:073314 --:------ 0016 001 00

18:730 is a pseudo address for your gateway (usb device).  You'll see it's real address in the reply.
01:073314 is your controller address (taken from the log sample above)

The controller should always respond - the last value of the RP from the controller is the received signal strength seen by the controller (hex) The value before RPis the received signal strength at the gateway (decimal)

Smaller values are better.  Depending on how far you are from your controller the values in your log sample above don't look too bad.

Grish-Lon, 2022-10-11:
Funny - I initially tried to purchase the device from there (local in the UK), but they were out of stock - so I got this one from Germany.  Incidentally, the one I received has a SMA connector for an antenna (also supplied).  The antenna is pretty "tall" for 868Mhz - about 3 inches long - maybe they started fixing antennas to them as the coil of wire was so-so in terms of working (compounding the crystal issue?)

I see that the signal strength that devices report are between 050-086 so I'm happy enough with the receive.

If I understand you right, the tuning is mainly for transmit.  Looks like that is working - here is the test result:
000 RQ --- 18:068639 01:073314 --:------ 0016 001 00
065 RP --- 01:073314 18:068639 --:------ 0016 002 0018

So a 65 signal strength?  That'll do fine!

The following is informational only/comment - I'm up and running - thank you for your help.

(BTW - am I suppose to close this ticket?? as its resolved :)

There are 18 devices (not including null and the nanoCUL) on the system and I've found them all:
TRVs (9)
Controller (1 :) )
Gateway (1) 
DHW sensor (1)
DTS92E (3)
BDR91 (3) - (Boiler relay, UFH relay, and I think the system pump relay)

Just have to work out which TRV is where, which DTS92E is where and what the BDR91s are for.

Assuming the BDR91 that controls the boiler is the one sending the "3EF0" codes and that those messages will be the firing times for the boiler - OFF="0000FF" and ON="00C8FF", but will have a look at ramses_cc board to see if there is any advice to see if that is the best method for determining firing times.

Also assuming gas is burned at the same rate always - its either burning or its not (not a combi boiler).  So if I can figure the volume/minute burned, I'll know how much is spent daily/weekly +/- some percent.

BTW - I've had this Honeywell system for 14 years.  It's been great to have individual room programs over the years, but always lacked the control and extra information that makes it really useful.  Glad I found this project (even if its been a while since you started it) - many thanks for your work!

Grish

ghoti57, 2022-10-11:
Yes close the issue.

The info on the  ramses_rf link will probably help you work out which bdr91 is which.

Check out using Home Assistant with evohome_cc (now ramses_cc) and ramses_rf for an easy way to collect and log information about your system.  Basic install info on evohome_cc wiki 

⁣Get BlueMail for Android ​

closed 2022-10-11 by Grish-Lon