ghoti57/evofw3 #31: nanoCUL not transmitting

trailro, 2022-10-16:
Hi all,

I seem to have an issue with evofw3 and my "german" nanoCUL. I flashed initially the latest version of evofw3 but it does not seem to work - the device "is" transmitting (i think) but no one ever responds back.

If I flash the calibrate branch, I can now talk to most devices - they all respond and quickly. Unfortunately, it's only the controller that does not want to respond back but only randomly to some commands. It doesn't even work for ping. But it all works fine for OTH and other devices.

Any suggestions as to what I can try next?

Thanks.

trailro, 2022-10-16:
nanoCUL atmega328p - seems to be somewhat a similar issue with https://github.com/ghoti57/evofw3/issues/6

ghoti57, 2022-10-16:
First of all forget the calibrate branch.  Anything in there is in the latest version.

How are you programming the nanoCUL?

Are you using my custom device definitions? What options are you using?

⁣Get BlueMail for Android ​

trailro, 2022-10-16:
Hi. Yes, I am using your custom definition for sw uart when uploading the sketch. For some reason, only using that branch I can send packets and receive responses. Using the master branch only allows me to send but no response whatsoever even after doing multiple freq calibrations.

Pins are set to nano. And baud to 112k. Standard bootloader.

ghoti57, 2022-10-16:
What do you see as serial output when you run the calibration?  In particular the last 20-30 lines.

trailro, 2022-10-17:
I will have to re-do the calibration. Do you want me to post the results with the master version or the current calibrate branch I have installed? Or both?

ghoti57, 2022-10-17:
Just forget the calibration branch exists

trailro, 2022-10-17:
Please see below full calibration using master branch:

14:19:00.857 -> ⸮# evofw3 0.7.1
14:19:05.687 -> 000 RQ --- 18:141092 10:256364 --:------ 0016 002 0017
14:19:10.103 -> 079  I --- 02:029004 --:------ 02:029004 22C9 024 0008660A28010108660A28010208660A28010307080A2801
14:19:11.348 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 002 0017
14:19:12.908 -> 070  I --- 01:256364 --:------ 01:256364 2309 030 0008660108340207080308340408340601F40701F40801F40907080A0708
14:19:12.908 -> 072  I --- 01:256364 --:------ 01:256364 30C9 024 0008390207F003083106070707065708071E0907DD0A07D9
14:19:14.323 -> 083 RP --- 10:056759 01:256364 --:------ 3EF0 009 000010020000032600
14:19:14.645 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 002 0017
14:19:19.595 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 002 0017
14:19:22.769 -> # !F F=2165b4
14:19:35.389 -> # !F Tune
14:19:35.418 -> # !c 0d 21 65 b4
14:19:35.418 -> # !c 0d 21 65 a4
14:19:36.572 -> 068  I --- 22:163985 --:------ 22:163985 0008 002 00C4
14:19:36.572 -> # !c 0d 21 65 94
14:19:37.576 -> 068  I --- 22:163985 --:------ 22:163985 0008 002 00C4
14:19:37.576 -> # !c 0d 21 65 84
14:20:14.495 -> 071 RQ --- 01:256364 10:056759 --:------ 3EF0 ???  * Invalid Manchester Code
14:20:14.495 -> # AA.5A.AA.95.56.69.96.5A.A6.6A.59.59.65.95.A5.56.55.AA.2A.
14:20:27.948 -> 075  I --- 04:092958 --:------ 01:256364 3150 002 0436 * Invalid Manchester Code
14:20:27.948 -> # A9.6A.A9.A9.96.65.A9.56.AA.95.56.69.96.5A.A5.A9.99.AA.AA.A6.AA.9A.A5.96.25.
14:20:33.053 -> 095  I --- --:------ --:------ --:------ ???? ???  * Invalid Manchester Code
14:20:33.089 -> # A9.6A.A9.A5.96.29.
14:21:08.364 -> 092  I --- 10:056759 --:------ 10:056759 3EF0 009 0032100A0000032600
14:21:08.364 -> # !c 0d 21 65 74
14:21:14.385 -> 060  I --- 01:256364 --:------ 01:256364 1F09 003 FF04BF
14:21:14.385 -> # !c 0d 21 65 64
14:21:14.418 -> 061  I --- 01:256364 --:------ 01:256364 000A 012 091001F40DAC0A1001F40DAC
14:21:14.418 -> # !c 0d 21 65 54
14:21:14.722 -> 060 RQ --- 01:256364 10:056759 --:------ 3EF0 001 00
14:21:14.722 -> # !c 0d 21 65 44
14:21:14.722 -> 092 RP --- 10:056759 01:256364 --:------ 3EF0 009 002B100A0000032600
14:21:14.758 -> # !c 0d 21 65 34
14:21:47.811 -> 094  I --- 22:163986 --:------ 22:163986 0008 002 0000
14:21:47.811 -> # !c 0d 21 65 24
14:21:51.801 -> 092  I --- 22:163986 --:------ 22:163986 0008 002 0000
14:21:51.833 -> # !c 0d 21 65 14
14:21:58.678 -> 091  I --- 10:056759 --:------ 10:056759 3EF0 009 000B10020000032600
14:21:58.678 -> # !c 0d 21 65 04
14:22:01.009 -> 078  I --- 02:029004 --:------ 01:256364 3150 002 0080
14:22:01.009 -> # !c 0d 21 64 f4
14:22:01.155 -> 080  I --- 02:029004 --:------ 01:256364 3150 002 0080
14:22:01.155 -> # !c 0d 21 64 e4
14:22:01.299 -> 078  I --- 02:029004 --:------ 01:256364 3150 002 0080
14:22:01.299 -> # !c 0d 21 64 d4
14:22:01.476 -> 078  I --- 02:029004 --:------ 01:256364 3150 002 0900
14:22:01.476 -> # !c 0d 21 64 c4
14:22:14.904 -> 061 RQ --- 01:256364 10:056759 --:------ 3EF0 001 00
14:22:14.941 -> # !c 0d 21 64 b4
14:22:14.941 -> 094 RP --- 10:056759 --:------ --:------ ???? ???  * Invalid Manchester Code
14:22:14.941 -> # A5.5A.A6.6A.59.59.65.95.AA.95.57.
14:22:48.008 -> 080  I --- 02:029004 --:------ --:------ ???? ???  * Invalid Manchester Code
14:22:48.008 -> # A9.6A.AA.6A.95.A9.9A.5A.AA.EA.
14:22:48.150 -> 079  I --- --:------ --:------ --:------ ???? ???  * Invalid Manchester Code
14:22:48.150 -> # A9.6A.AA.6F.
14:23:15.859 -> 061  I --- 01:256364 --:------ 01:256364 1F09 003 FF04BF
14:23:15.895 -> # !c 0d 21 64 a4
14:23:15.954 -> 062 RQ --- 01:256364 --:------ --:------ ???? ???  * Invalid Manchester Code
14:23:15.954 -> # AA.5A.AA.95.56.69.96.5A.A6.EA.
14:25:16.359 -> 060 RQ --- --:------ --:------ --:------ ???? ???  * Invalid Manchester Code
14:25:16.359 -> # AA.5A.AA.D5.
14:27:15.571 -> # !c 0d 21 64 ac
14:31:15.256 -> # !c 0d 21 64 b0
14:35:14.905 -> # !c 0d 21 64 b2
14:36:02.549 -> 079  I --- 02:029004 --:------ 01:256364 3150 002 007E
14:36:02.549 -> # !c 0d 21 64 b1
14:36:02.686 -> 076  I --- 02:029004 --:------ --:------ ???? ???  * Invalid Manchester Code
14:36:02.686 -> # A9.6A.AA.6A.95.A9.9A.5A.AA.9F.
[... 64 lines cut ...]
15:02:24.068 -> 060 RQ --- 01:256364 10:056759 --:------ 3EF0 001 00
15:02:24.068 -> 091 RP --- 10:056759 01:256364 --:------ 3EF0 009 000010000000032600
15:02:33.368 -> 091  I --- 10:056759 --:------ 10:056759 3EF0 009 000010020000032600
15:02:40.249 -> # !F F=2165b5
15:02:44.840 -> # !F Saved F=2165b5

Now, having flashed this branch, I can no longer receive responses from any of the devices:
15:03:24.268 -> 077 RQ --- 01:256364 10:056759 --:------ 3EF0 001 00
15:03:24.314 -> 092 RP --- 10:056759 01:256364 --:------ 3EF0 009 000010020000032600
15:03:27.018 -> 092  I --- 04:092960 --:------ 04:092960 30C9 003 000839
15:03:45.820 -> 082  I --- 01:256364 --:------ 01:256364 1F09 003 FF04BF
15:03:45.878 -> 081  I --- 01:256364 --:------ 01:256364 000A 048 001001F40DAC010001F40DAC021001F40DAC031001F40DAC040001F40DAC061001F40DAC071001F40DAC081001F40DAC
15:03:45.878 -> 081  I --- 01:256364 --:------ 01:256364 000A 012 091001F40DAC0A1001F40DAC
15:03:53.475 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 002 0017
15:04:03.117 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 002 0017
15:04:05.759 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 002 0017
15:04:12.220 -> 000 RQ --- 18:141092 01:256364 --:------ 1F09 001 00
15:04:18.765 -> 080  I --- 02:029004 --:------ 02:029004 22C9 024 0008660A28010108660A28010208660A28010301F40A2801
15:04:24.466 -> 075 RQ --- 01:256364 10:056759 --:------ 3EF0 001 00
15:04:24.523 -> 089 RP --- 10:056759 01:256364 --:------ 3EF0 009 000010020000032600
15:04:28.347 -> 000 RQ --- 18:141092 10:056759 --:------ 1F09 001 00

Previously:
2022-10-17T06:56:36.701422 000 RQ --- 18:141092 10:056759 --:------ 1290 001 00
2022-10-17T06:56:36.729984 077 RP --- 10:056759 18:141092 --:------ 1290 003 007FFF

trailro, 2022-10-17:
As for flashing, this is what I used:

[image]

ghoti57, 2022-10-17:
The tuning has done nothing significant, the value it’s chosen is 003% from the ideal.

 

In case the nvram data is causing an issue reset it with the command

!ER

 

Power cycle and check the frequency value with 

!F

It should be 21656A

 

Then try a TX operation again.

 

 

I think the guy who reported this issue

<https://github.com/ghoti57/evofw3/issues/30>

Bought the same stick and once he programmed it correctly had no problems.

He’s also using a MAC

 

You must have the correct programming options because otherwise you’d be getting no responses at all.

 

Peter

 

 <https://user-images.githubusercontent.com/38461924/196174089-43fb73d7-1dad-4436-83be-6283e37dff25.png> e tuning has achieved nt

trailro, 2022-10-17:
Hi Peter,

Just did that and unfortunately no luck in getting any responses :( 

what's strange is that I was getting some using the <not mentioned again> calibrate branch. so i'm wondering what's changed since than that could cause this reaction.

18:25:26.571 -> # evofw3 0.7.1
18:25:32.261 -> # !F F=21656a
18:25:44.981 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 001 00
18:25:50.066 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 001 00
18:25:55.948 -> 000 RQ --- 18:141092 10:056759 --:------ 10E0 001 00
18:26:07.045 -> 066 RQ --- 01:256364 10:056759 --:------ 3EF0 001 00
18:26:15.754 -> 067  I --- 01:256364 --:------ 01:256364 1F09 003 FF04BF
18:26:15.776 -> 067  I --- 01:256364 --:------ 01:256364 2309 030 0008660108340201F40308340408340601F40701F40801F40908340A01F4
18:26:15.776 -> 067  I --- 01:256364 --:------ 01:256364 30C9 024 0008BD0207E90308410607370706CF0807270907C70A07C1
18:26:21.201 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 001 00
18:26:25.760 -> 000 RQ --- 18:141092 01:256364 --:------ 0016 001 00

djfanatix, 2022-11-11:
Did you manage to fix this device?
I have the same issues: not transmitting, device 'crashing' after a few minutes,..

I tried evofw2, calibrate and latest.
I did autotune and !ER, but no major improvements.
I think it is a hardware issue, so maybe I just return it and buy a proper one

trailro, 2022-11-11:
Unfortunately, no. I have it running with the calibrate branch (which I shouldn't be using but at least I get some data out of it). Otherwise no luck whatsoever. I also think it might be a hardware issue - but it's strange that it somehow works when i use an older fw version. Hard to say. I already ordered another stick - hopefully the new one will work.

sciorty, 2023-04-21:
@trailro i do have the same problem of a CC1101 + Arruino UNO apparently not transmitting. Dis you solve in any way?

noud-github, 2024-01-10:
Looks like using the "calibrate" branch made my nanocul TX come alive to, 
unfortunately it looks like the nano version cul are the only once you can buy at the moment,

ghoti57, 2024-01-10:
I have stock of https://indalo-tech.onlineweb.shop/product/ssm-d2 but am unable to dispatch until the end of January.

I am hiding the stock level because I don't want people to order and then get frustrated when I don't post it out.

If you haven't done so already request an update on the page linked above and I'll notify you when you can place an order.

closed 2024-01-10 by ghoti57

noud-github, 2024-01-10:
@ghoti57 does that ship to the EU? (had some VAT issues before so asking it 😉

looks like 0.6.17 introduced the issue on my current stick on 0.6.16 I can send  on 0.6.17 it is broken 😞

ghoti57, 2024-01-10:
It is available in the EU but you will have to pay VAT and a handling charge on import.  See the delivery FAQ.  I reduce the declared value by British VAT.  So you don't pay tax on tax but am not VAT registered so it is difficult for me to stop duty paid.