ghoti57/evofw3 #32: Strange behavior on Remeha Elga Ace heatpump after activating Nanocul (evofw3) with Home Assistant (ramses_cc)

henkvdt, 2022-10-20:
**Description of setup:**

Evohome controller, connected to Remeha Elga Ace heatpump with R8810A1018 (Opentherm module) 
Firmware version: WiFi module: 02.00.17.00, Application module: 02.00.19.33

Home Assistant config:

```
ramses_cc:
  serial_port: 
    port_name: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0
    baudrate: 57600  #  default: 115200
  packet_log: 
    file_name: packet.log
    rotate_backups: 30
```

Nanocul from this supplier: https://schlauhaus.biz/en/product/nanocul-868/
Firmware: 0.7.1

**Problem**

After setup of the nanocul and software, all the sensors and entities became visible in Home Assistant. However, the heatpump started showing strange behaviour. Even though there is a heat demand from the Evohome to the heatpump, and output/return watertemp are below set point, every 6-10 minutes the heatpump stops heating for about 2 minutes. After this, the heatpump turns back on. The display on the heatpump doesn't show anything abnormal or error codes.

As soon as I disabled the ramses_cc plugin in Home Assistant, the heatpump started operating normally again. This can be seen in the energy diagram as attached. There was constant heatdemand most of the time between 11.35 and 13.50, but the heatpump was switching on and off. The peak around 13.40 was caused by setting the Evohome to 25 degrees to force a very high setpoint, but here the heatpump also stopped way before the setpoint was reached. 

I disabled the plugin at 13.50. The Evohome send heat demand between 13.50 - 14.15 and 14.40 - 15.08. Between these timestamps, there is clear correlation between heat demand and power usage. Between 14.56 and 15.08 there was heat demand, but the heatpump reached it's setpoint. Therefore only the circulation pump was active which causes the lower power usage.

[image]

Outdoor temperature is between 10-15 degrees celcius, so defreeze cycles are not relevant.

It seems like the nanocul is interfering the messages between the Evohome and heatpump (connected through R8810A1018) Could this be caused by active probing?

ghoti57, 2022-10-20:
This is an issue for ramses_cc

Evofw3 only transmits messages when instructed by application software in its host system.  In this case ramses_cc in HA

⁣Get BlueMail for Android ​

WJM73, 2023-01-17:
Hello, just reading this tread and searching for an application whitin home assistant tot connect to the Elga Ace 6kW heatpump of possible in combination with bluetooth, also connected with Honeywell Evohome with the only difference, that I use the on/off module with a second one to change over for cooling.

The part that interessest me is how you are connected with the Elga, is this direct through bluetooth or Honewell Evohome.

if I can hellp, let me know

ghoti57, 2023-01-17:
See my comment on Oct 20 2022.

This is not the place to ask questions like this.  

Look for ramses_cc which is the interface to evohome in home assistant.

⁣Get BlueMail for Android ​

closed 2023-02-14 by ghoti57