ghoti57/evofw3 #38: Evofw3 in Domoticz

roblomq, 2023-03-10:
Short question. Is it possible to use this evofw3 in Domoticz combined with a nanoCul? If yes is there some guide on how to do this?

ghoti57, 2023-03-11:
Evofw3 will work on a nanocul.  I recommend the later versions based on an atmega32u4 rather than the atm328p devices, they will be much more effective at capturing RAMSES messages.

I can't suggest any guides for using it with domoticz, search the forums for references to HGI80.

Alternatively if you are not committed to domoticz yet, look at Home Assistant and the custom component ramses_cc.
<https://github.com/zxdavb/ramses_cc>

As an alternative to a nanocul I sell this device pre-loaded with evofw3
<https://indalo-tech.onlineweb.shop/>

Peter

⁣Get BlueMail for Android ​

roblomq, 2023-03-11:
I already use the Domoticz plugin that uses the web API which works fine but what I'm missing there is the best demand of the zone. So I was hoping when using the nanocul this heat demand would be visible. So I already bought your nanocul 😅 

I've found the instruction below.

https://www.domoticz.com/wiki/Evohome#Adding_Evohome_via_usb_device_.28HGI80_or_equivalent.29

This gives me 18 devices in Domoticz but they are all boiler related, I don't get the zone information from the nanocul.
What isn't completely clear is how to obtain the Controller ID, I now used the crc from the Evohome device but not sure it's correct.

ghoti57, 2023-03-11:
The controller ID is visible in all the serial messages.  

If domoticz doesn't allow you to see them plug the SSM into a pc and open the COM port that will appear with a serial terminal device (e.g. putty).

The messages are printed as they are received in a fixed format that has 3 ID fields.  Anything like 01:xxxxx is the controller.  

That's about all the help I can give because I have no first hand knowledge of domoticz.

If you're only using Domoticz for the evohome system you should still consider switching to HA.  There's a lot of ongoing development.

If you're using the SSM in either system the add-ons that go out to the Honeywell (Residio) web interface are irrelevant on both systems.

roblomq, 2023-03-11:
I get the following:

```
049 RQ --- 01224380 10:037143 --:------ 3EF0 001 00
107 RP --- 10:037143 01:224380 --:------ 3EF0 009 0000100200FF032C00
076  I --- 10:037143 --:------ 10:037143 3EF0 009 0000100200FF032C00
092  I --- --:------ --:------ 12:225222 0008 002 006E
092  I --- --:------ --:------ 12:225222 0008 002 006E
092  I --- --:------ --:------ 12:225222 0008 002 006E
072  I --- 22:161953 --:------ 22:161953 0008 002 0052
071  I --- 22:161953 --:------ 22:161953 0008 002 0052
062  I --- 04:152535 --:------ 04:152535 30C9 003 000771
062  I --- 04:150765 --:------ 01:224380 3150 002 0400
052 RQ --- 01:224380 10:037143 --:------ 3EF0 001 00
067 RP --- 10:037143 01:224380 --:------ 3EF0 009 0000100200FF032C00
068  I --- 10:037143 --:------ 10:037143 1FD4 003 002EF1
067  I --- --:------ --:------ 20:000227 31D9 003 000000
051  I --- 01:224380 --:------ 01:224380 1F09 003 FF0541
051  I --- 01:224380 --:------ 01:224380 000A 048 001001F40DAC011001F40DAC021001F40DAC031001F40DAC041001F40DAC051001F40DAC061001F40DAC071001F40DAC
091  I --- 20:006313 --:------ 20:006313 31DA 029 00EF007FFFEFEF7FFF7FFF7FFF7FFFF000EF0101000000EFEF7FFF7FFF
067  I --- 04:150759 --:------ 04:150759 30C9 003 00092A
```
So then is 224380 the ID?

roblomq, 2023-03-11:
I used the 224380 in the "Controller ID (hex)" field of the Domoticz hardware settings but nothing changes still no zones and only 19

[image]

Some of the log info from Domoticz
```
2023-03-11 15:56:53.090 Status: evohome serial: Opening serial port: /dev/serial/by-id/usb-SparkFun_evofw3_atmega32u4-if00@115200
2023-03-11 15:57:18.852 Evohome USB: General/Percentage (Relative modulation level)
2023-03-11 15:57:18.891 Evohome USB: Temp (Control Setpoint)
2023-03-11 15:57:18.843 Status: evohome: opentherm bridge detected, ID:0x289117
2023-03-11 15:57:34.028 Evohome USB: General/Percentage (Relative modulation level)
2023-03-11 15:57:34.068 Evohome USB: Temp (Control Setpoint)
2023-03-11 15:57:49.776 Evohome USB: General/Percentage (Relative modulation level)
2023-03-11 15:57:49.808 Evohome USB: Lighting 2 (FlameOn)
2023-03-11 15:57:49.846 Evohome USB: Temp (Control Setpoint)
2023-03-11 16:06:58.428 Evohome USB: General/Custom Sensor (Application fault flags)
2023-03-11 16:06:58.479 Evohome USB: General/Custom Sensor (OEM fault code)
2023-03-11 16:06:59.413 Evohome USB: General/Percentage (Relative modulation level)
2023-03-11 16:07:00.395 Evohome USB: General/Pressure (CH Water Pressure)
2023-03-11 16:07:04.412 Evohome USB: Temp (Outside Temperature)
```

ghoti57, 2023-05-12:
The controller ID in the formatted message is 01:224380
These DECIMAL fields are a representation of a 24 bit value 0x076c7c

Try this hex value as the controller ID.

roblomq, 2023-05-13:
That seems to be working. Thank you!

closed 2023-05-13 by roblomq