ghoti57/evofw3 #42: Error messages

RBL-Hormann-ALK, 2023-11-11:
I receive the error messages below when I use a nanocul usb stick in Domoticz. There are several devices discovered and works as expected but was hoping the error messages can be eliminated.

2```
023-11-11 19:06:33.218 Error: Evohome USB: evohome: unknown command 12c8
2023-11-11 19:06:56.216 Error: Evohome USB: evohome: unknown command 1298
2023-11-11 19:07:13.475 Error: Evohome USB: evohome: unknown command 31da
2023-11-11 19:08:13.561 Error: Evohome USB: evohome: invalid message structure - possible corrupt message
2023-11-11 19:08:35.352 Error: Evohome USB: evohome: unknown command 1298
2023-11-11 19:08:36.922 Error: Evohome USB: evohome: unknown command 1fd4
2023-11-11 19:08:41.773 Error: Evohome USB: evohome: unknown command 2e10
2023-11-11 19:09:29.766 Error: Evohome USB: evohome: unknown command 31e0
2023-11-11 19:09:37.971 Error: Evohome USB: evohome: unknown command 2e10
2023-11-11 19:09:42.831 Error: Evohome USB: evohome: unknown command 31d9
2023-11-11 19:10:06.858 Error: Evohome USB: evohome: unknown command 31e0
2023-11-11 19:10:19.075 Error: Evohome
```

ghoti57, 2023-11-11:
This is an issue to take up with the Domoticz team.

Evofw3 just captures the radio messages and reports then to a host. It does not generate visible error messages.

closed 2023-11-11 by ghoti57

RBL-Hormann-ALK, 2023-11-12:
Ok that's clear, thanks. I already tried to find the person that made the Evohome code in Domoticz because nobody responds to my questions. (And yes I know in HA it works better.. )

RBL-Hormann-ALK, 2023-11-12:
I hope you been help me with the following. In Domoticz I also have several devices that display zone information like the below example.

Heating | Relay | Set Level: 62 %: 124

What is the 124 and is the 62% the demand or something?

ghoti57, 2023-11-12:
I'm sorry I have no experience with Domoticz

⁣Get BlueMail for Android ​